robertinant/EnergiaNG #1: MSP430F5529 UniFlash 3.4.1 Compatibility

rei-vilo, 2016-06-29:
The latest release of UniFlash 3.4.1 updated the firmware for the MSP430F5529 and MSP432. 

Does Energia plan to share the same release of the firmwares?

rei-vilo, 2016-06-29:
No more **Tools > Update Programmer** menu — See #3

robertinant, 2016-06-29:
Energia will use whatever firmware is available in DSLite. I will try to keep DSLite as up to date as possible as much as I can to keep in sync with what is available in CCS and Uniflash. However, if the user has an older version of either of these then there will be a firmware mismatch and an upgrade/downgrade will happen when moving from one to the other.

closed 2016-06-29 by robertinant

rei-vilo, 2016-06-29:
DSLite performed fine for the MSP432 but failed to update the firmware for the MSP430F5529. I had to use Energia 17 and upload the firmware again.