robertinant/EnergiaNG #2: MSP432 UniFlash 3.4.1 Compatibility

rei-vilo, 2016-06-29:
The latest release of UniFlash 3.4.1 updated the firmware for the MSP430F5529 and MSP432.

Does Energia plan to share the same release of the firmwares?

rei-vilo, 2016-06-29:
No more **Tools > Update Programmer** menu — See #3

robertinant, 2016-06-29:
I just updated DSLite to version 6.2.1.1624 and that should match what is in uniflash. Also, all OS'es should not have the same firmware included for all emulators so no more upgrading/downgrading when you move your board from on OS to the other. Unfortunately tools can not be updated separate from the cores and the the core package dependent on it has to change revision for the change to take effect. I will do this starting with MSP432 that will be updated later today and then soon after for the other core packages.

closed 2016-06-29 by robertinant