robertinant/EnergiaNG #6: Splash Screen Still with 17 Release

rei-vilo, 2016-06-29:
[image]

RickKimball, 2016-06-29:
Also a problem on linux

rei-vilo, 2016-06-29:
Suggestion...

[image]

robertinant, 2016-06-29:
Yes, the version number is displayed in both the title bar as well as in the top left corner of the splash screen. Both of these are generated from the version numbers in the IDE. So no need for an explicit version number and going to remove. In fact, if you can send me that splash screen in the right size then I'll use that.

closed 2016-07-07 by robertinant