robertinant/EnergiaNG #17: Unable to build Energia on Mac OS X

rei-vilo, 2016-08-21:
I'm launching the build with 

```
$ cd ~/GitHub/Energia/EnergiaNG/build
$ ant -buildfile build-energia.xml clean diet
```

The process fails with this error:

>
compile:
    [mkdir] Created dir: /Users/ReiVilo/GitHub/Energia/EnergiaNG/arduino-core/bin
    [javac] Compiling 139 source files to /Users/ReiVilo/GitHub/Energia/EnergiaNG/arduino-core/bin
    [javac] javac: invalid target release: 1.8
    [javac] Usage: javac <options> <source files>
    [javac] use -help for a list of possible options

BUILD FAILED
/Users/ReiVilo/GitHub/Energia/EnergiaNG/build/build-energia.xml:115: The following error occurred while executing this line:
/Users/ReiVilo/GitHub/Energia/EnergiaNG/build/build-energia.xml:92: The following error occurred while executing this line:
/Users/ReiVilo/GitHub/Energia/EnergiaNG/build/build-energia.xml:139: The following error occurred while executing this line:
/Users/ReiVilo/GitHub/Energia/EnergiaNG/arduino-core/build.xml:41: Compile failed; see the compiler error output for details.
>

More details:

```
$ java -version
java version "1.7.0_80"
Java(TM) SE Runtime Environment (build 1.7.0_80-b15)
Java HotSpot(TM) 64-Bit Server VM (build 24.80-b11, mixed mode)
```

```
$ ant -version
Apache Ant(TM) version 1.9.7 compiled on April 9 2016
```

rei-vilo, 2016-08-21:
Looks like JDK 1.8 is compulsory.

See http://www.oracle.com/technetwork/java/javase/downloads/jdk8-downloads-2133151.html

closed 2016-08-21 by rei-vilo

robertinant, 2016-08-22:
What triggered the need for executing with sudo? Can you paste the failure that you see when you build without sudo?

rei-vilo, 2016-08-22:
Access to one file.

```
/Users/ReiVilo/GitHub/Energia/EnergiaNG/arduino-core/build.xml:12: Unable to delete file /Users/ReiVilo/GitHub/Energia/EnergiaNG/arduino-core/bin/cc/arduino/Compiler$1.class
```