robertinant/EnergiaNG #22: Choose between DSlite and LM4flash?

szymekrak, 2016-09-08:
I have downloaded the new version of Energia and noticed that lm4flash is no longer used as a tool for uploading programs to a Stellaris/Tiva C board, but instead "DSLite" is there. LM4flash is still with the toolchain's bin folder, but it can not be used in the IDE - in the menu there are only two copies of "dslite" and "rf2500". I liked the fact that lm4flash was free software(free as in free speech), but the new tool is of unknown license or at least I can not find it anywhere.
So, will there be option to use lm4flash again or is it gone forever from Energia? Sorry if there is an option for this that I missed.

RickKimball, 2016-09-15:
See the changes I made to the platform.txt and boards.txt

I've got it using openocd now but you can also use lm4flash
https://github.com/RickKimball/tivac-core/commit/542bdc15215da539081784747e76591bdc8e0ab5

szymekrak, 2016-09-17:
Dear @RickKimball,
I have just tried the core from Your repository and it seems to work just fine, thanks! I think I will stick with this if there won't be any problems. As I also use Arduino, now I don't need to have two copies of almost the same IDE.