robertinant/EnergiaNG #40: Workshop — Compiling Against MSP432 Red Fails With Hard-Coded Path

rei-vilo, 2018-06-04:
Compiling against the MSP432 Red LaunchPad fails with hard-coded path `fork/exec /bin` for `arm-none-eabi-g++`.

This happens on Windows 10, even when the path to Energia contains no spaces. It also happens on macOS.

```
fork/exec /bin/arm-none-eabi-g++: no such file or directory
Error compiling for board RED LaunchPad w/ msp432 EMT (48MHz).
```

```
Energia: 1.6.10E18 (Mac OS X), Board: "RED LaunchPad w/ msp432 EMT (48MHz)"

/Applications/IDE/Energia-18.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware "/Applications/IDE/Energia-18.app/Contents/Java/hardware" -hardware "~/Library/Energia15/packages" -tools "/Applications/IDE/Energia-18.app/Contents/Java/tools-builder" -tools "/Applications/IDE/Energia-18.app/Contents/Java/hardware/tools/msp430" -tools "~/Library/Energia15/packages" -built-in-libraries "/Applications/IDE/Energia-18.app/Contents/Java/libraries" -libraries "~/Documents/Projets/Energia/libraries" -fqbn=energia:msp432r:MSP-EXP432P401RR -ide-version=10610 -build-path "/var/folders/px/cyfvtr757lqg0yp_cv9j79jh0000gn/T/buildd38424d699e7bf83e76e15fec77a88fe.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/var/folders/px/cyfvtr757lqg0yp_cv9j79jh0000gn/T/arduino_modified_sketch_911746/sketch_apr14a.ino"
/Applications/IDE/Energia-18.app/Contents/Java/arduino-builder -compile -logger=machine -hardware "/Applications/IDE/Energia-18.app/Contents/Java/hardware" -hardware "~/Library/Energia15/packages" -tools "/Applications/IDE/Energia-18.app/Contents/Java/tools-builder" -tools "/Applications/IDE/Energia-18.app/Contents/Java/hardware/tools/msp430" -tools "~/Library/Energia15/packages" -built-in-libraries "/Applications/IDE/Energia-18.app/Contents/Java/libraries" -libraries "~/Documents/Projets/Energia/libraries" -fqbn=energia:msp432r:MSP-EXP432P401RR -ide-version=10610 -build-path "/var/folders/px/cyfvtr757lqg0yp_cv9j79jh0000gn/T/buildd38424d699e7bf83e76e15fec77a88fe.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/var/folders/px/cyfvtr757lqg0yp_cv9j79jh0000gn/T/arduino_modified_sketch_911746/sketch_apr14a.ino"
Using board 'MSP-EXP432P401RR' from platform in folder: ~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1
Using core 'msp432r' from platform in folder: ~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1
Warning: Board energia:c2000:lptms320f28069 doesn't define a 'build.board' preference. Auto-set to: C2000_LPTMS320F28069
Warning: Board energia:c2000:lptms320f28377s doesn't define a 'build.board' preference. Auto-set to: C2000_LPTMS320F28377S
Warning: Board energia:c2000:lptms320f28027 doesn't define a 'build.board' preference. Auto-set to: C2000_LPTMS320F28027
Warning: Board energia:cc26xx:CC2650STK_BLE doesn't define a 'build.board' preference. Auto-set to: CC26XX_CC2650STK_BLE
Detecting libraries used...
"/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mabi=aapcs -g -Dxdc_target_types__=gnu/targets/arm/std.h -Dxdc_target_name__=M4F -Dxdc_cfg__xheader__="configPkg/package/cfg/energia_pm4fg.h" -DBOARD_MSP_EXP432P401R -Dxdc__nolocalstring=1 -D__MSP432P401R__ -DCORE_VERSION=561 -c -g -Os -w -ffunction-sections -fdata-sections -fno-threadsafe-statics -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m4 -mthumb -DF_CPU=48000000L -DARDUINO=10610 -DENERGIA=10610 -DENERGIA_MSP_EXP432P401R -DENERGIA_ARCH_MSP432R "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/cores/msp432r/ti/runtime/wiring/" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/cores/msp432r/ti/runtime/wiring/msp432" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/kernel/tirtos/packages/ti/sysbios/posix" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/kernel/tirtos/packages" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/source" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/energia" -D_DEFAULT_SOURCE "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/source/ti/devices/msp432p4xx/driverlib" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/source/ti/devices/msp432p4xx/inc" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/source/ti/devices/msp432p4xx/" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/source/third_party/CMSIS/Include" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/cores/msp432r/ti/runtime/wiring/" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/cores/msp432r/ti/runtime/wiring/msp432" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/kernel/tirtos/packages/ti/sysbios/posix" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/kernel/tirtos/packages" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/source" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/energia" -D_DEFAULT_SOURCE "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/source/ti/devices/msp432p4xx/driverlib" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/source/ti/devices/msp432p4xx/inc" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/source/ti/devices/msp432p4xx/" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/source/third_party/CMSIS/Include" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/cores/msp432r" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/variants/MSP_EXP432P401R" "/var/folders/px/cyfvtr757lqg0yp_cv9j79jh0000gn/T/buildd38424d699e7bf83e76e15fec77a88fe.tmp/sketch/sketch_apr14a.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mabi=aapcs -g -Dxdc_target_types__=gnu/targets/arm/std.h -Dxdc_target_name__=M4F -Dxdc_cfg__xheader__="configPkg/package/cfg/energia_pm4fg.h" -DBOARD_MSP_EXP432P401R -Dxdc__nolocalstring=1 -D__MSP432P401R__ -DCORE_VERSION=561 -c -g -Os -w -ffunction-sections -fdata-sections -fno-threadsafe-statics -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m4 -mthumb -DF_CPU=48000000L -DARDUINO=10610 -DENERGIA=10610 -DENERGIA_MSP_EXP432P401R -DENERGIA_ARCH_MSP432R "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/cores/msp432r/ti/runtime/wiring/" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/cores/msp432r/ti/runtime/wiring/msp432" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/kernel/tirtos/packages/ti/sysbios/posix" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/kernel/tirtos/packages" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/source" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/energia" -D_DEFAULT_SOURCE "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/source/ti/devices/msp432p4xx/driverlib" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/source/ti/devices/msp432p4xx/inc" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/source/ti/devices/msp432p4xx/" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/source/third_party/CMSIS/Include" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/cores/msp432r/ti/runtime/wiring/" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/cores/msp432r/ti/runtime/wiring/msp432" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/kernel/tirtos/packages/ti/sysbios/posix" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/kernel/tirtos/packages" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/source" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/energia" -D_DEFAULT_SOURCE "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/source/ti/devices/msp432p4xx/driverlib" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/source/ti/devices/msp432p4xx/inc" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/source/ti/devices/msp432p4xx/" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/system/source/third_party/CMSIS/Include" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/cores/msp432r" "-I~/Library/Energia15/packages/energia/hardware/msp432r/5.6.1/variants/MSP_EXP432P401R" "/var/folders/px/cyfvtr757lqg0yp_cv9j79jh0000gn/T/buildd38424d699e7bf83e76e15fec77a88fe.tmp/sketch/sketch_apr14a.ino.cpp" -o "/var/folders/px/cyfvtr757lqg0yp_cv9j79jh0000gn/T/buildd38424d699e7bf83e76e15fec77a88fe.tmp/preproc/ctags_target_for_gcc_minus_e.cpp"
fork/exec /bin/arm-none-eabi-g++: no such file or directory
Error compiling for board RED LaunchPad w/ msp432 EMT (48MHz).
```

robertinant, 2018-06-04:
Which version of the core was this?

rei-vilo, 2018-06-05:
MSP432 EMT RED Boards version 5.6.1