robertinant/EnergiaNG #41: Workshop — Allow Path with Spaces for Energia Installation

rei-vilo, 2018-06-04:
According to [How to set up Energia on Windows](http://energia.nu/guide/guide_windows/)

> Energia (MT) uses a different method for building Sketches for the MSP-EXP432P401R. Because of a limitation in this method you must install Energia in a location without spaces.

Allow the installation of Energia MT in a folder which path contains spaces and special characters.

robertinant, 2018-06-04:
I have not seen issues with spaces in the path for MT targets such as MSP-EXP432P401R. What is the error produced when there is a space in the Energia path?

rei-vilo, 2018-06-05:
The path with space issue concerns the Energia installation, not the targets.