robertinant/EnergiaNG #43: Workshop — Issue with PWM on MSP430F2955 + EduBP-MK2

rei-vilo, 2018-06-05:
The following basic sketch sometimes runs and sometimes freezes on the MSP430F2955 LaunchPad and Educational BossterPack-MK2.

```
// Define variables and constants
#define redLED 39
#define greenLED 38
#define blueLED 37

// Add setup code
void setup()
{
    Serial.begin(9600);
    Serial.println("*** RGB LED");

    pinMode(redLED, OUTPUT);
    pinMode(greenLED, OUTPUT);
    pinMode(blueLED, OUTPUT);
}

// Add loop code
void loop()
{
    for (uint16_t i = 0; i<256; i++) {
        Serial.print("\r Red-Green   ");
        Serial.print(i, DEC);
        analogWrite(redLED, 255-i);
        analogWrite(greenLED, i);
        delay(10);
    }

    for (uint16_t i = 0; i<256; i++) {
        Serial.print("\r Green-Blue  ");
        Serial.print(i, DEC);
        analogWrite(greenLED, 255-i);
        analogWrite(blueLED, i);
        delay(10);
    }

    for (uint16_t i = 0; i<256; i++) {
        Serial.print("\r Blue-Red   ");
        Serial.print(i, DEC);
        analogWrite(blueLED, 255-i);
        analogWrite(redLED, i);
        delay(10);
    }
}
```

StefanSch, 2018-06-08:
Does it freeze directly after loading or also if it is running while.

rei-vilo, 2018-06-08:
`*** RGB LED` is sent to the console, so it freezes after.

StefanSch, 2018-06-08:
but only sometimes ?
Can you try the same with a faster baud rate, e.g. 115k?

StefanSch, 2018-06-08:
Can you also check this line
#define redLED 39
the device only has 38 pins

rei-vilo, 2018-06-11:
Same issue with `Serial.begin(115200);`, but this time nothing is sent to the console, the issue occurs before sending the first `Serial.println("*** RGB LED");`.

`#define redLED 39` refers to the LaunchPad pin 39, as defined in `pins_energia.h` [line 164](https://github.com/energia/msp430-lg-core/blob/c80ffd6498fb8a8228a7eac802e7c68458e6cc40/variants/MSP-EXP430F5529LP/pins_energia.h#L164)

```
static const uint8_t P1_5 = 38;
```

StefanSch, 2018-06-11:
The link to pins_energia.h goes to the MSP430F5529 but the title says MSP430F2955 ???

rei-vilo, 2018-06-11:
Sorry, I was talking about the **MSP430F5529**.