robertinant/EnergiaNG #47: SPI other than the default one doesn't work on MSP430FR2433 launchpad

saint-shark, 2018-11-10:
I'm trying to use the UCA0 for SPI on MSP430FR2433 by using the SPI.setmodule(10); before SPI.begin(), but I'm not getting any data on the SPI lines. The default SPI on this launchpad is UCA1 and it works. 

After the above thing didn't work, I tried changing the core files: /home/username/.energia15/packages/energia/hardware/msp430/1.0.5/variants/MSP-EXP430FR2433LP/pins_energia.h

In this file I changed: (from line 47 to 50)
`static const uint8_t SS11     = 8;   /* P2.1 */
 static const uint8_t SCK11    = 7;   /* P2.3 */
 static const uint8_t MOSI11   = 15;  /* P2.4 aka SIMO */
 static const uint8_t MISO11   = 14;  /* P2.5 aka SOMI */`

to:
`static const uint8_t SS11     = 8;       /* P2.1 */
 static const uint8_t SCK11    = 5;     //P1_6
 static const uint8_t MOSI11   = 4;    //P1_4 aka SIMO */
 static const uint8_t MISO11   = 3;    //P1_5 aka SOMI */`
 
And changed:
`#define DEFAULT_SPI 11`

to:
`#define DEFAULT_SPI 10`

But this also didn't work.

StefanSch, 2018-11-12:
I think the mix if change max the problem

Try to add this lines to the pins_energia.h file:
static const uint8_t SS10     = 8;   /* P2.1 */
static const uint8_t SCK10    = 5;   /* P1.6 */
static const uint8_t MOSI10   = 4;   /* P1.4 aka SIMO */
static const uint8_t MISO10   = 3;   /* P1.5 aka SOMI */
#define SPISCK10_SET_MODE  (PORT_SELECTION0)
#define SPIMOSI10_SET_MODE (PORT_SELECTION0)
#define SPIMISO10_SET_MODE (PORT_SELECTION0)

and use the setModule(10); 

This allows you to switch between the 2 USCI modules.

Note: to make you fix work you need to use SS10, SCK10, .....
Note2: i did not test this - so there might be some risk that it does not work. In this case let me know.

saint-shark, 2018-11-12:
Thanks @StefanSch, adding those lines worked for me. 
Is this going to be added in the future release? And similar to this, many boards have this issue(I tried on MSP430FR5969 too), that  multiple SPI, I2C and UART don't work. I think only minor changes are required to add these functionalities. I can make these changes and drop in a pull request(if allowed).

StefanSch, 2018-11-13:
Hi,
Contribution is always welcome, pushing updates to branch and triggering a merge request should be allowed to you. If not just let me know.
UART and I2C should be in quite good shape but for SPI i just started to add this as it is not supported in Arduino so far.

StefanSch, 2019-02-04:
fixed in energia-lg with issue_79 merge request

closed 2019-02-04 by StefanSch